echolevel/open-amiga-sampler #7: Add phono preamp?

kikendo, 2020-07-30:
Hey guys, interesting project here!

I just thought it'd be neat to have a phono preamp in here to be able to sample straight from a turntable without the need of a phono preamp in the middle.
Since there's really not many ways to improve sampler functionality in the Amiga, I thought this'd be a neat idea.

Feel free to throw it in the trash xD

echolevel, 2020-07-30:
Thanks! I do like this idea, and I'm wondering how best it could be done. I guess it would need a built-in RIAA filter, and I'm not sure whether there are single-chip solutions for that (at least, ones that can run on ~5v). I've seen lots of DIY phono preamps that are built like tanks and powered by enormous switching PSUs but that might be a red herring - the designers and hobbyists building those things might have other requirements in mind, in terms of SNR or power output.

But this also reminds me that I haven't got around to just sticking a turntable right into the OAS yet. I'm sure it'll sound a bit grim without the RIAA curve compensation, but I'm curious! It would be nice to save on a power outlet and free up some phono cable runs by being able to go turntable direct to sampler...

kikendo, 2020-07-30:
You managed to change the label now, cool! ;)
And yeah just a wild idea, but if you can pull it off, it'd definitely give it an edge over any other sampler ever made.

j0hnlittle, 2020-08-08:
Maybe as a starting point, there is this application note from Maxim: [USB-POWERED HIGH PRECISION HI-FI PHONO PREAMP](https://www.maximintegrated.com/en/design/technical-documents/app-notes/1/1931.html)

> This design idea shows a 3.3V powered phone preamplifier with RIAA equalization curve. The advantage is the low power, low voltage single supply solution. It is done with a MAX4478, a quad, low-noise, rail-to-rail op amp.

[image]

> A small linear regulator, the MAX8877, regulates the noisy +5V from the USB supply down to +3.3V

[image]