echolevel/open-amiga-sampler #11: Gerber file

Milesy, 2020-10-10:
Hi, I would like to print a pcb using pcb but having issues trying to generate the Gerber file using Kicad.

Could you please supply the required config or a Gerber file?

Many thanks

Chris

Milesy, 2020-10-10:
Note - for the classic edition, not surface mount

abrugsch, 2020-10-10:
By default KiCAD doesn't generate drill files so I expect that's your problem, so after you generate Gerber's from the default options click the generate drill files button and again default options should be fine. Alternatively if you use OSH Park you can just submit the kicad_pcb file and they take care of the rest

abrugsch, 2020-10-13:
@Milesy here's a plot of the classic gerbers. i've checked them with OSH park, they should be fine in any of the PCB fabs
[OAS-Gerbers.zip](https://github.com/echolevel/open-amiga-sampler/files/5370184/OAS-Gerbers.zip)