echolevel/open-amiga-sampler #15: okay to sell?

smartroad, 2021-05-24:
Hi! I wanted to make one of these for myself but have to buy several PCBs. I was wondering if it would be okay to sell the remainder, maybe with the components as a kit?

abrugsch, 2021-05-24:
Which one is it? The SMD version hasn't been tested as working yet. @echolevel got one working but it was very very quiet even with gain turned right up.

I will be investigating shortly now my makerspace is open again.

If it's the regular through hole variant then I don't see a problem selling off the spares.

smartroad, 2021-05-24:
It would be the through hole. I tend to do TH as I have a hot air wand and while I do some limited SMD soldering that way, I prefer through hole :)

Ruxton, 2021-05-26:
based on https://github.com/echolevel/open-amiga-sampler/wiki/Availability i took it as gospel that it was OK to build up and sell.  made 5 boards, sold 2 bare, kept 2 for myself and traded one with a mate. A few people in the local Amiga group expressed interest in obtaining some pre-built ones, so I figured if this run worked well I'd order a few more boards and do some assemblies for them.

mnstrmnch, 2021-05-26:
Thank you for checking, and yes please go ahead and do as you wish with the through-hole version. It would be cool to see your results too - we're aware of people having made their own but (as far as I know) haven't seen any in the wild yet :)

smartroad, 2021-05-26:
Great thanks! I have the board on order (be a couple of weeks as I am cheap for postage lol). Modified the board a little to move the e-caps to the underside on the board (hope that's okay too!). Got parts on order so hopefully should have one built in a few weeks.

echolevel, 2021-05-27:
Good stuff! Might be worth paraphrasing a response I gave this week to someone who emailed about making some to sell on eBay (also in Australia, but not @Ruxton!)

My initial run has stalled temporarily, partly because I'm flat-out with work but also because I've got no ventilated space to do a batch of soldering in until I move house, hopefully soon. The units in that run are all spoken for - I've kept a spreadsheet that's as fair as I can make it which lists usernames of everyone who's shown an interest via YouTube/Instagram/Twitter replies, sorted in the chronological order of those replies. The idea is for me to contact those people via those platforms when the units are ready and see if they're still interested.

If somebody else makes a batch before I finish mine, I'm sure a bunch of those people will be very pleased to get one quicker. I'm all for that: the point of this project is to get them into people's hands and help them avoid eBay price-gouging on retro kit! That'll free up some slots on my list. And even if it frees up ALL the slots on my list (ie the market is saturated and I can no longer shift my units), so be it :) 

So it follows that if anyone reading this _does_ want to make up a batch to sell, let me know in this thread and I'll do my best to inform the people on my list that there are units available if they don't want to wait for mine.

P.S. I have no idea what constitutes market saturation for this product. It could be 100 units, it could be a dozen. Retro computing is weird at the moment - lots of enthusiastic newcomers with disposable income, which is great, but I detect a trend of people buying a peripheral for every hole, as it were, regardless of whether they intend to use it very much or at all - more out of completeness. That's absolutely fine, I'm not bothered what people do with their own stuff; I'm just saying it's why I struggle to estimate how many people in the world will want one of these. 

So as long as no prospective manufacturers go nuts and order e.g. 500 boards and find themselves out of pocket (I think it's safe to say that would be a wild overestimation), I think a handful of us doing a dozen or couple of dozen units will work out nicely!

smartroad, 2021-05-27:
I've only ordered 5 pcbs so will have 4 spares. Also ordered enoigh parts for all as well. I might pop it onto Tindie just to ease payments etc for me. Not really interested in making a profit other than to cover my time and testing!

I have an idea for a case as well and maybe a logo (I might be becoming too invested here haha!)

smartroad, 2021-05-28:
Once I have my first one built and validated, I can pop on there and see if anyone would like any of the others?

echolevel, 2021-05-28:
It's a good place to start if looking for interested buyers, for sure - just check the rules (probably pinned in the main channel) for protocol on offering things for sale, but I _think_ you can informally announce in the Amiga channel that you're selling. Lots of people in there are comfortable building their own too, so you might get interest in blank PCBs.

abrugsch, 2021-05-28:
@smartroad please be aware that on RMC discord there's a privilege to sell things that's RMC patron only...
RMC might not be the best place to organise that

smartroad, 2021-05-28:
@abrugsch that's rubbish :( well I'll put mine up on Tindie and people can find it there (or not!) I might pop a message on the hackaday article as well to say I have a few to sell.

smartroad, 2021-05-28:
While I wait for the PCBs to arrive I have been desinging a case and logo for the sampler. Here is the first version mocked up:
[image]

It is designed to just clip together so no need to fasteners hopefully. I miss measured the gain pot so this versions knob is to short. Also the fit is a little bit to tight causing the top and bottom parts to flex a little.

Next version will have increased fit between the clips holding it together and the knob will be increased in length.

Hope people like! ☺

smartroad, 2021-05-29:
Next version:
[image]

I'll also make a version that doesn't need dual/multi extrusion, although the text wouldn't be another colour of course.

smartroad, 2021-05-30:
I thought about making a label for the back of the case for basic instructions and info about the device/project. What do people think of the wording here:
[image]

Ruxton, 2021-06-04:
@smartroad  i really like your label, only grip is the spacing around the QR code vs. everything else.

smartroad, 2021-06-04:
@Ruxton cheers! Yeah I am not that happy with it. I think I'll see if QR codes need to be black on white and test if it works inverted 🤔 That would allow easier fitting without making it much smaller (it's only about 20mm square!)

smartroad, 2021-06-04:
@Ruxton 
[image]
Tweaked QR code - inverting colours doesn't work on QR codes it seems LOL

echolevel, 2021-06-06:
Looks good! Big Mackie Big Knob energy 👍🏼

smartroad, 2021-06-10:
PCBs have arrived so will be building one at the weekend to test!

smartroad, 2021-06-13:
https://youtu.be/jzzcCFz7-dA

It lives!!!

echolevel, 2021-06-15:
Nice one! Great to see one out in the wild!

smartroad, 2021-06-16:
Planning to record a proper video over the weekend doing a bit of an overview. For selling, I am just waiting on an order of resistors as I didn't have any 330R's.

smartroad, 2021-06-19:
Overview video is currently rendering and I hope to have it up soon! I have a PAINFULLY slow internet connection so it may be a while even after rendering to get it fully uploaded and ready to go. I'll provide a link once I have it uploaded :)

smartroad, 2021-06-20:
Preview of video for those interested :) plan to release tomorrow (1pm BST)

echolevel, 2021-06-20:
Excellent! 

The only thing I'd be picky about - purely in terms of technical pedantry, and not suggesting you make any changes - is the fact that the OAS' maximum sample rate is actually around 50Khz! Obviously there's an enormous pile of caveats involved, and you definitely wouldn't be able to run an Amiga program's display - let alone spectrum analysis - while sampling at those rates, but it can be done. Even in Protracker the practical upper limit is around 29Khz (A-3/B#3), and that's what you use if you've got chipmem to burn and you want the highest possible quality hi-hats :) 

Needless to say, these intricacies are way more about the Amiga itself (and PAL/NTSC video timing, and the CIA chip) than about OAS and its chip specifically. [More details here](https://github.com/echolevel/open-amiga-sampler/wiki/Appendix-A:-Sample-rates-deep-dive). 

Out of interest, I wonder what the maximum possible sample rate is on one of these A/D chips? Probably quite high (albeit still at 8bit). Some time when I can be arsed, I'll test it on my Teensy parallel-port-emulator jig.

smartroad, 2021-06-20:
Cheers! I will update the video and include a link to the page. I noticed I missed a couple of other bits off I wanted to say so have added those in as well.

Max sample rate, if I am reading the data sheet right, is about 400K or so (I made it 460K but I don't think I am including all of the timing delays)

echolevel, 2021-06-20:
Haha nice - I knew it was something preposterous and ultimately limited only by its internal silicon witchcraft, but couldn't recall what sort of ballpark it was in. 

If you ever get the chance btw, it's a nice little exercise to try sampling a >16bit and >44.1Khz signal at the highest possible rate using Megalosound or similar and then comparing it to the source. If you choose a source that's sympathetic towards the reduction to 8 bits - e.g. pretty hot, a musically dense mix, not much dynamic range, definitely no subtle fades - you can get surprisingly impressive results. 

Of course it's no more than that, an exercise, since you can't then play those samples back at the original frequency without also switching to the crazy-hi-frequency screenmode, but it's fun all the same.

smartroad, 2021-06-21:
Updated video on full release: https://youtu.be/Eji0Wg25V3g

echolevel, 2021-06-22:
Nice one! Just shared it on [Twitter](https://twitter.com/echolevel/status/1407248513407279108) so you might get some queries from people interesting in buying the other four...

pawelomega, 2021-09-06:
Im also finished my AOS. Im very happy. Working like a charm 
[image]

hayesey, 2021-09-30:
Does anyone have a spare bare PCB they'd be willing to sell rather than me getting another batch made up?

pawelomega, 2021-10-02:
sorry. my 4 pcs all goes to Amiga friends.

smartroad, 2021-10-02:
> Does anyone have a spare bare PCB they'd be willing to sell rather than me getting another batch made up?

I think I may have a couple still here, if you are still in need I'll have a look! Have all the components as well I think!