echolevel/open-amiga-sampler #16: BOM more information about parts

pawelomega, 2021-06-15:
Hi. Im trying to build my fisrt own sampler but i need more info about used parts as:

1x 330ohm resistor - how many watts? 5,10,50,100 W
5x 1kohm resistors - also same as upper

1x 47uf capacitor - how many volts, 16,25,50,100 V and type - electrolytic capacitor or tantalum
2x 10uf capacitors - also same as upper
2x 0.1uf capacitors -also same as upper

thanks in advance for more important info for me.

echolevel, 2021-06-15:
Hi Pawel, 

Capacitor voltage ratings are usually upper limits, beyond which the capacitor might burn out. Similarly, power (in Watts) ratings for resistors are upper limits. This circuit is extremely low voltage and low power, so for example a capacitor rated at 50V will be fine, but so will a capacitor rated at 63V or 100V - as long as the capacitance value is correct, and the capacitor is the correct physical size to fit into the circuit. 

Similarly it shouldn't matter whether a resistor (in this circuit) is rated for 0.25W or 2W, since the circuit should never be handling that much power. We're looking at 5V from the Amiga's parallel port, which is enough top power the A/D chip and the op amp, and 24 mA (milliamps), which gives us a wattage of around 0.12, so neither the voltage nor the wattage in this circuit will come close to the limits of most of the resistors and capacitors that you'll find when searching for the BOM's specs! 

As for electrolytic vs tantalum: electrolytics are cheaper and I don't think tantalum caps would make any noticeable improvement, given the relatively lo-fi audio quality we're dealing with here :)

Happy to hear from anyone else who might be better informed than me, but generally speaking as long as the components' W/V ratings exceed the throughput in the circuit, then variations in those ratings should be fine.

pawelomega, 2021-06-15:
Nice. PCBs are ordering. It's time to complet parts. Thanks !

pawelomega, 2021-08-05:
Hello again. I have problem to find 2x 10uf capacitors ceramic in my country.  Only tantal or electrolitic  Can i swap with electrolitic 10uF 50V ? but need know polarisation if possible.

gwEm303, 2021-08-18:
I would make C1 tantalum, the others can be electrolytic. Make sure the positive terminal is connected to the opamp.

Circuit all runs at 5v or below, so 6v or above capacitors will be fine.

1/8 watt resistors will be fine.