echolevel/open-amiga-sampler #21: Better anti-aliasing filter

gwEm303, 2021-08-18:
Using the unused U2b we may implement this 18dB/octave anti-aliasing filter:

https://www.dropbox.com/s/mzd1efro0kk7ato/OAS%20cascaded%20antialiasing%20filter.jpg?dl=0

jnlundberg, 2024-08-12:
photo of soldered pcb would be great

gwEm303, 2024-08-12:
sorry, I can't take one of those at the moment

jnlundberg, 2024-08-13:
"The 1nF can be connected on the solder side of the PCB, between pins 4 and 5 of U2b." 

-should this instead be pins 5 and 6 of U2b, as in the schematic?

butterrsoft, 2024-10-30:
This mod replaces the previous single-gang pot, is that correct? When implemented, does the new pot still dial in variable gain, and the filter follows? Or does it just adjust the filter, and the gain is fixed?

And no, jnlundberg, the cap is not shown between pins 5 and 6. It goes between 4 and 5 (4 being ground).

gwEm303, 2024-10-30:
> This mod replaces the previous single-gang pot, is that correct? When implemented, does the new pot still dial in variable gain, and the filter follows? Or does it just adjust the filter, and the [...]

that’s right!

butterrsoft, 2024-10-30:
Err, sorry, that answer is a little confusing! So the new pot replaces the old one, and still adjusts the sample gain, yes?

gwEm303, 2024-10-30:
:-D

no, you end up with two pots : one for the gain and one for the anti-aliasing filter

butterrsoft, 2024-10-30:
So the old pot connects in after U2A pin 1 but before the leftmost 5k6 resistor, yes?

gwEm303, 2024-10-30:
the pot to adjust the filter is a dual gang pot and only does the filter. one gang of it connects after the first 5K6 and the other gang of it connects after the second 5K6

to control the gain there is a 100K log pot which is in the feedback loop of U2A (see “improved input stage”)

butterrsoft, 2024-10-30:
Righto, thanks for the pointers :)

jnlundberg, 2024-10-30:
Hey guys,  I never got this to work. Please let me know if succeed. Would be nice with a picture. And an updated schematic with all the mods. Not sure how to trouble shoot it. Pcb should be ok, I ordered with a midi interface pcb and it worked out fine.

butterrsoft, 2024-10-30:
I'm marking up a new schematic to make an smd pcb from. Not sure if the 3rd order filter will fit, but I can share the schematic when done.

jnlundberg, 2024-10-30:
I got parts to do a few more of this through hole so it'd be great if I van use the schematic with it

butterrsoft, 2024-10-31:
OAS schematics:
with 1st order filter - https://fastupload.io/f11ed2d0d8b00241
with 3rd order filter - https://fastupload.io/8083af4fe8b074d2

Both use the V.20 conventional input stage

jnlundberg, 2024-10-31:
Amazing, thnx a lot!

echolevel, 2024-10-31:
Good stuff, everyone! @butterrsoft - would you mind reporting back on whether you have success with either/both circuits? If you can confirm success, we can add the PDFs to the repo (if you give us permission, of course). Thanks!

gwEm303, 2024-11-24:
It’s just a low pass filter, not a high/low pass filter.

I would like to add a high pass filter to the circuit to have both options, but that’s not done.

The intention of the low pass filter is to remove whine and aliasing noise. so it’s possible the filter is working well!

butterrsoft, 2024-11-25:
Ahh, that's interesting. In which case, yes, the LPF works amazingly well! In this case i was able to mitigate the problem with 'better' source equipment, but that's not always going to be the case.

rain-fade, 2025-06-15:
> OAS schematics: with 1st order filter - https://fastupload.io/6788f0d7c0f750b3 with 3rd order filter - https://fastupload.io/49db4cdfce478cad
> 
> Both use the v2.0 conventional input stage from gwEm303 :)

sadly both of these links resolve to scammy ads now.

butterrsoft, 2025-06-15:
> sadly both of these links resolve to scammy ads now.

[Schematic_OAS_v2.0-with-1st-order-filter.pdf](https://github.com/user-attachments/files/20748028/Schematic_OAS_v2.0-with-1st-order-filter.pdf)
[Schematic_OAS_v2.0-with-3rd-order-filter.pdf](https://github.com/user-attachments/files/20748027/Schematic_OAS_v2.0-with-3rd-order-filter.pdf)